brainexe/node-metawear #1: Add Gyro/Accelerometer support

brainexe, 2015-09-19:


AlexDmr, 2015-12-15:
Indeed, I would love it too :)

onnobos, 2016-01-01:
Great work so far. Any idea when the Accelerometer part is ready?

brainexe, 2016-01-02:
The base implementation of the gyroscope seems to work for me. See examples/gyro.js

Yesterday i started the implementation of the accelerometer sensor in branch "accelerometer". See #6  
But it's far away from stable status. It might me usable end of next week

onnobos, 2016-01-02:
The gyro is working for me and especially after your last updates it runs much quicker and I get the information directly :-). So far I have created a small node application using socket.io and rgraph for a graph. I will be using this to identify jumps for my volleyball team. The accelerometer will be part of plan. 

Thanks for your work so far. BTW Do you re-engineer from the java application from metawear? I'm really interesting how you came up with your solution.

brainexe, 2016-01-04:
Hey @onnobos, yes i used the Android Java code as base. 
The two most important files are:
- https://github.com/mbientlab/Metawear-AndroidAPI/blob/master/library/src/main/java/com/mbientlab/metawear/impl/DefaultMetaWearBoard.java
- https://github.com/mbientlab/Metawear-AndroidAPI/blob/master/example/src/main/java/com/mbientlab/metawear/example/MainActivity.java

brainexe, 2016-01-27:
Sorry for delay, but now I merged the first working implementation of the accelerometer into master #6 
You can try the test script in examples/accelerometer.js

But this is only a rough implementation without possibility to change configuration yet
Open tasks:
 - implement correct scaling of result ([reference values](https://github.com/mbientlab/Metawear-AndroidAPI/blob/04a18a8b3eb79669a3b3f96951953ff1991c26d6/library/src/main/java/com/mbientlab/metawear/module/Bmi160Accelerometer.java))
 - configurable MOTION_CONFIG (dataRate/accRange)
 - configurable TAP_CONFIG
 - configurable ORIENT_CONFIG
 - configurable FLAT_CONFIG
 - configurable LOW_HIGH_G_CONFIG
Feel free to create a pull request for some missing part. I will create further issues for detailed information. If you need help, just ask

closed 2016-02-04 by brainexe