brainexe/node-metawear #4: Problem running the example

lgleasain, 2015-10-16:
I'm on a chromebook pixel running ubuntu trusty 14.4 under crouton which should have access to the full bluetooth stack etc..   After installing all of the libraries etc. I tried running the all example and it hangs without any output on the screen (no logs etc).  Any ideas about what might be happening?

siddrajput, 2015-10-16:
Did you try running it in debug mode? Like this:

For debug output  DEBUG="noble-device" node examples/all.js

lgleasain, 2015-12-08:
Figured it out.  Got it working on OSX,  crouton in Ubuntu and on a rasberry Pi.

closed 2015-12-08 by lgleasain

brainexe, 2015-12-08:
Hey @lgleasain what was the issue? Did you made any changes? 
Some useful information to add to the readme file?

lgleasain, 2015-12-08:
My big issue was that I needed to turn on debugging to see that something was happening and then run the logging via the project directory....  Give me a day or two and I'll put in a pull request for the readme along with a blog post.