brainexe/node-metawear #5: Getting error cannot read property on for undefined.

lgleasain, 2015-12-07:
This is on the latest version of rasberrian for the Rasberry Pi.

sudo DEBUG="noble-device" node node_modules/node-metawear/examples/all.js 
  noble-device start +0ms
were connected!
/home/pi/code/mbient/node_project/node_modules/node-metawear/src/registers/data-processing.js:38
    this.emitter.on([MODULE_OPCODE, NOTIFY], function(buffer) {
                ^

TypeError: Cannot read property 'on' of undefined
    at new DataProcessing (/home/pi/code/mbient/node_project/node_modules/node-metawear/src/registers/data-processing.js:38:17)
    at /home/pi/code/mbient/node_project/node_modules/node-metawear/examples/all.js:19:31
    at null._onTimeout (/home/pi/code/mbient/node_project/node_modules/node-metawear/src/device.js:43:17)
    at Timer.listOnTimeout (timers.js:92:15)

lgleasain, 2015-12-08:
Ok  tracked this down.  It appears to be a problem with the data processor.  If I comment this out in the all.js file it works....not sure what the issue is.

brainexe, 2015-12-08:
Hey @lgleasain, i'll fix the root cause of the issue later. The data-logger is only a simple dummy implementation and was not tested fully from my side.

brainexe, 2015-12-23:
fixed in 1325a49c3b3

closed 2015-12-23 by brainexe