brainexe/node-metawear #8: Implement config for accRange/dataRate

brainexe, 2016-01-27:
I added already the possibility to set dataRate/accRange, but they are not taken into account yet...

Bennoo, 2016-02-04:
Hello, do you know where I can start looking for the issue?
I would like to have 100 Hz but it is not taken into account..
Tx

brainexe, 2016-02-06:
I extended the examples/accelerometer.js script by two optional CLI parameters. The first is the sampling rate in HZ and the second one the accRange.

In general you can pass the accRange/dataRate now...but i found one issue, related to sampling rates  below 6.25hz and above 1600hz. In this case i don't get any value.

6.25hz and below -> **FAIL**...see commands below
12.5hz -> working fine with entries each 75ms-90ms as expected...no bigger gaps on my device
25hz  -> working fine with 35ms-45ms gaps
50hz -> working fine with 15ms-30ms gaps
100hz  -> working fine with 5ms-15ms gaps
200hz -> working fine
400hz -> working fine
800hz -> fine with 0ms-5ms
1600hz and above -> **FAIL**see commands below

**6.25hz**
  noble-device  +4s send LOGGING <Buffer 0b 0b 00>
  noble-device  +1ms send LOGGING <Buffer 0b 01>
  noble-device  +1ms send ACCELEROMETER <Buffer 03 03 05 06>
  noble-device  +1ms send ACCELEROMETER <Buffer 03 04 01>
  noble-device  +0ms send ACCELEROMETER <Buffer 03 02 01 00>
  noble-device  +0ms send ACCELEROMETER <Buffer 03 0d 04 0a>
  noble-device  +0ms send ACCELEROMETER <Buffer 03 0a 00 14 14 14>
  noble-device  +0ms send ACCELEROMETER <Buffer 03 03 **24** 03> <-** DATA_CONFIG command**
  noble-device  +1ms send ACCELEROMETER <Buffer 03 07 07 30 81 0b c0>
  noble-device  +0ms send ACCELEROMETER <Buffer 03 01 01>

**1600hz**
  noble-device  +6s send LOGGING <Buffer 0b 0b f8>
  noble-device  +1ms send LOGGING <Buffer 0b 01>
  noble-device  +2ms send ACCELEROMETER <Buffer 03 03 05 06>
  noble-device  +0ms send ACCELEROMETER <Buffer 03 04 01>
  noble-device  +0ms send ACCELEROMETER <Buffer 03 02 01 00>
  noble-device  +0ms send ACCELEROMETER <Buffer 03 0d 04 0a>
  noble-device  +0ms send ACCELEROMETER <Buffer 03 0a 00 14 14 14>
  noble-device  +0ms send ACCELEROMETER <Buffer 03 03 **2d** 03> <- **DATA_CONFIG command**
  noble-device  +1ms send ACCELEROMETER <Buffer 03 07 07 30 81 0b c0>
  noble-device  +0ms send ACCELEROMETER <Buffer 03 01 01>

Try to use data rates between 12.5 and 800hz for now only.

closed 2016-02-12 by brainexe