brainexe/node-metawear #14: Question On Creating onStep function in Register

yowakita, 2016-02-24:
Hello, first of all thank you for creating this module, it has been very useful for me. Pardon my lack of knowledge because a lot of this is still new to me, but I would love to learn more.

I am attempting to create an onStep function for the accelerometer that would utilize the constants you have in the accelerometer.js register file, 
```
STEP_DETECTOR_INTERRUPT_ENABLE = 0x17;
STEP_DETECTOR_CONFIG    = 0x18;
STEP_DETECTOR_INTERRUPT = 0x19;
STEP_COUNTER_DATA       = 0x1a;
STEP_COUNTER_RESET      = 0x1b;
```

Attempting to emulate the code you have now on the onChange function, I wrote a function
```
Accelerometer.prototype.onStep = function (callback) {
    this.device.emitter.on( [MODULE_OPCODE, STEP_DETECTOR_INTERRUPT], function(buffer){
        callback();
    });
}
```
Ands added to the `start` function 
```
    buffer = new Buffer(3);
    buffer[0] = MODULE_OPCODE;
    buffer[1] = STEP_DETECTOR_CONFIG;
    buffer[2] = 0x00;
    this.device.send(buffer);
```

When I write into the example accelerometer.js file,
 ```
accelerometer.onStep(function(){
     console.log("Test");
 })
```
I don't get an error but I also don't get anything console log. I am certain I am not understanding how buffers and MODULE_OPCODE work. Could you please help?

brainexe, 2016-02-27:
I quickly checked the Android SDK source code and it seems you have to extend the STEP_DETECTOR_CONFIG call by the "stepCnt" (by default 0x15, 0x3)
->
https://github.com/mbientlab/Metawear-AndroidAPI/blob/04a18a8b3eb79669a3b3f96951953ff1991c26d6/library/src/main/java/com/mbientlab/metawear/impl/DefaultMetaWearBoard.java#L5680

Additionally you might send a STEP_DETECTOR_INTERRUPT_ENABLE request as well. But you have to check
->
https://github.com/mbientlab/Metawear-AndroidAPI/blob/04a18a8b3eb79669a3b3f96951953ff1991c26d6/library/src/main/java/com/mbientlab/metawear/impl/DefaultMetaWearBoard.java#L5697

And maybe it's useful to call your script in debug mode to check the whole request/response stream.

For further help it would be useful if you could send a link to the full code. (fork repo + create new feature branch)

yowakita, 2016-03-11:
When you say
`I quickly checked the Android SDK source code and it seems you have to extend the STEP_DETECTOR_CONFIG call by the "stepCnt" (by default 0x15, 0x3)`
Do you mean sending a buffer like 
```
    buffer = new Buffer(3);
    buffer[0] = MODULE_OPCODE;
    buffer[1] = 0x15;
    buffer[2] = 0x03;
    this.device.send(buffer);
```
into the `start` function and then adding 
```
    buffer = new Buffer(3);
    buffer[0] = MODULE_OPCODE;
    buffer[1] = STEP_DETECTOR_INTERRUPT;
    buffer[2] = 0x1;
    this.device.send(buffer);
```
to the `enableNotifications` function

I don't have anything of substance at the moment, but I would like to create a pedometer app.

I forked your repo and made some changes that I hope are in the right direction in the accelerometer register file.

https://github.com/yowakita/node-metawear/tree/stepCount

 Any help on what to do to get it working would make me super happy. I have spent a lot of time looking over the documentation and patterns of your library and the metawear API but I feel like I am not advanced enough to know what I'm doing wrong exactly. Thank you!

brainexe, 2016-03-12:
Hey @yowakita, i pushed my version of the step counter a few seconds ago: #17 
You can call ./examples/stepCounter.js
This will print a dot for each step and the sum of all steps after each 5 seconds. The number is provided by internal metawear counter and can be resetted via Accelerometer.resetStepCounter()

In general I added 3 calls to enable the step counter:
STEP_DETECTOR_CONFIG
STEP_DETECTOR_INTERRUPT_ENABLE
STEP_DETECTOR_INTERRUPT

Is it enough, or do you need more functionality or information?

yowakita, 2016-03-14:
Thats great @brainexe ! I tried out your example and it seems to work very well so far. I'm going to fiddle with the example and study your code a bit this week, thanks so much for your help!

closed 2016-03-29 by brainexe