brainexe/node-metawear #15: Windows Support

JpEncausse, 2016-02-25:
Hi,
Any plan for NodeJS / Windows10 support ?

Thanks !

alanhortz, 2016-02-27:
@JpEncausse Do you mean Windows 10 iot core ?

JpEncausse, 2016-02-27:
No Windows 10 desktop edition.
I 'd like to put Metaware as a core component of SARAH (http://sarah.encausse.net) 
SARAH works on NodeJS (server) and C# (client)  and it would be great to be able to get MetaWare data.

For instance this PoC use a Table with a Windows 10 on an Intel NUC
https://www.youtube.com/watch?v=VErQZcZBivM

But I have to use a mobile phone as a proxy :-(

brainexe, 2016-02-27:
sorry, i don't have any windows device for checking for compatibility...

But in general this libary relies on [sandeepmistry/noble](https://github.com/sandeepmistry/noble)
Have you checked  [this introduction](https://github.com/sandeepmistry/noble#windows)?

brainexe, 2016-03-13:
Any news here?

alanhortz, 2016-11-11:
@JpEncausse Do you have any update ? 

I understand that you would let the node process access to the MetaWear, did you workaround that by using the CPP bindings in C# directly ?