brainexe/node-metawear #16: cannot run examples

felipecocco, 2016-03-01:
noble-device start +0ms
discovered device  unknown
/Users/felipecocco/Downloads/node-metawear-master/examples/accelerometer.js:14
    device.onSetup(function(error) {
           ^
TypeError: undefined is not a function
    at /Users/felipecocco/Downloads/node-metawear-master/examples/accelerometer.js:14:12
    at EventEmitter.onDiscover (/Users/felipecocco/Downloads/node-metawear-master/node_modules/noble-device/lib/util.js:76:9)
    at EventEmitter.emit (events.js:107:17)
    at Noble.constructor.onDiscover (/Users/felipecocco/Downloads/node-metawear-master/node_modules/noble-device/lib/util.js:30:29)
    at Noble.emit (events.js:107:17)
    at Noble.onDiscover (/Users/felipecocco/Downloads/node-metawear-master/node_modules/noble-device/node_modules/noble/lib/noble.js:152:10)
    at emit (events.js:118:17)
    at null.<anonymous> (/Users/felipecocco/Downloads/node-metawear-master/node_modules/noble-device/node_modules/noble/lib/mac/yosemite.js:188:12)
    at /Users/felipecocco/Downloads/node-metawear-master/node_modules/noble-device/node_modules/noble/lib/mac/uuid-to-address.js:22:5
    at tryParseBuffer (/Users/felipecocco/Downloads/node-metawear-master/node_modules/noble-device/node_modules/noble/node_modules/bplist-parser/bplistParser.js:25:5)

brainexe, 2016-03-01:
sorry, the API was changed slightly in the last PR. I just fixed the example scripts in c91c847

felipecocco, 2016-03-01:
Works like a charm -- thank you so much! Awesome work.

closed 2016-03-01 by felipecocco