brainexe/node-metawear #18: Streaming throughput slows after a few seconds

felipecocco, 2016-03-13:
I'm running your example accelerometer code on a MBP, and it streams wonderfully at first. However, after a few seconds, the stream noticeably slows down and I get maybe two updates a second, instead of the flurry of updates that I get at the beginning. If disconnect and reset, I again get the great stream throughput but always get the slowdown afterwards.

brainexe, 2016-03-13:
I can't verify your issue on my MetaWear Pro device.

I ran the examples/accellerometer.js script for ~5 minutes and the output stream is constant. (at 50hz)
Are you using the most recent firmware on your device?

Can you please start the script in debug mode and post the output directly after starting the script and after the slowdown.

> DEBUG="noble-device" node examples/accelerometer.js
> discovered device  f9:d5:4f:...
> were connected!
> Start accelerometer with 50hz ang +-2g
>   noble-device  +3s send LOGGING <Buffer 0b 0b 00>
>   noble-device  +2ms send LOGGING <Buffer 0b 01 01>
>   noble-device  +1ms send ACCELEROMETER <Buffer 03 03 27 03>
>   noble-device  +0ms send ACCELEROMETER <Buffer 03 04 01>
>   noble-device  +0ms send ACCELEROMETER <Buffer 03 02 01 00>
>   noble-device  +0ms send ACCELEROMETER <Buffer 03 0d 04 0a>
>   noble-device  +0ms send ACCELEROMETER <Buffer 03 0a 00 14 14 14>
>   noble-device  +0ms send ACCELEROMETER <Buffer 03 03 27 03>
>   noble-device  +0ms send ACCELEROMETER <Buffer 03 07 07 30 81 0b c0>
>   noble-device  +1ms send ACCELEROMETER <Buffer 03 01 01>
> x: 0.0380859375 		y: -0.18890380859375 		z: -1.01226806640625
>   noble-device  **+28ms** received ACCELEROMETER 4 <Buffer 03 04 70 02 e9 f3 37 bf>
> x: 0.0389404296875 		y: -0.18768310546875 		z: -1.0125732421875
>   noble-device  **+13ms** received ACCELEROMETER 4 <Buffer 03 04 7e 02 fd f3 32 bf>
>   noble-device  **+14ms** received AMBIENT_LIGHT 3 <Buffer 14 03 00 00 00 00>
> x: 0.03887939453125 		y: -0.187255859375 		z: -1.01287841796875
>   noble-device  **+15ms** received ACCELEROMETER 4 <Buffer 03 04 7d 02 04 f4 2d bf>
> x: 0.03887939453125 		y: -0.1861572265625 		z: -1.01214599609375
>   noble-device  **+16ms** received ACCELEROMETER 4 <Buffer 03 04 7d 02 16 f4 39 bf>
> x: 0.03875732421875 		y: -0.1876220703125 		z: -1.013916015625
>   noble-device  **+29ms** received ACCELEROMETER 4 <Buffer 03 04 7b 02 fe f3 1c bf>
> 
> ...after 5 minutes:
>
>   noble-device  **+15ms** received ACCELEROMETER 4 <Buffer 03 04 85 02 ec f3 1e bf>
> x: 0.0367431640625 		y: -0.187744140625 		z: -1.01361083984375
>   noble-device  **+15ms** received ACCELEROMETER 4 <Buffer 03 04 5a 02 fc f3 21 bf>
> x: 0.0384521484375 		y: -0.1883544921875 		z: -1.01275634765625
>   noble-device  **+45ms** received ACCELEROMETER 4 <Buffer 03 04 76 02 f2 f3 2f bf>
> x: 0.038330078125 		y: -0.1873779296875 		z: -1.013916015625
>   noble-device  **+1ms** received ACCELEROMETER 4 <Buffer 03 04 74 02 02 f4 1c bf>
> x: 0.03863525390625 		y: -0.18792724609375 		z: -1.0126953125
>   noble-device ** +14ms** received ACCELEROMETER 4 <Buffer 03 04 79 02 f9 f3 30 bf>
> x: 0.0379638671875 		y: -0.18756103515625 		z: -1.01165771484375
>   noble-device  **+30ms** received ACCELEROMETER 4 <Buffer 03 04 6e 02 ff f3 41 bf>
> x: 0.03924560546875 		y: -0.18817138671875 		z: -1.01324462890625

felipecocco, 2016-03-13:
Yes, I'm running the latest firmware on my Metawear. Here's a little clip showing you exactly the behavior I'm seeing. This happens very time I run it. Also, unless I do a soft reset of the metawear device, it stays in that "slowed down" mode from the outset if I run the script again.

streamable.com/9uu6

yowakita, 2016-04-27:
@felipecocco : I noticed the same thing with my macbook pro (late 2014 model.) I am wondering if it has to do with one of Node-Metawear's dependent libraries and their interaction with OSX because I tested some code on my MBP and got the same ~1000ms choppiness, and then tested the same code on my raspberry pi B+ and an Ubuntu desktop and didn't run into any choppiness issues.

alanhortz, 2016-04-29:
Just for the record ... I had exactly the same problem with my MBP (El Capitan 10.11.3), no issue on Yocto. Until now I gave up digging into the problem on the MBP to focus on the logging.

alanhortz, 2016-11-11:
Does anybody have update on this ?

adebiyi422, 2016-11-16:
I am having the same issue on an Ubuntu desktop.

alanhortz, 2016-11-17:
@felipecocco : Which version FW,HW and Model number for your device ? We have updated the dependencies. Do you confirm you still encounter the same problem ?

felipecocco, 2016-11-20:
@alanhortz I seem to encounter a different problem now.

The stream no longer halts after a while; now from the start I seem to get output that is roughly on a +130ms delay followed by 0/1ms delays at regular intervals.

When I first run the command to test out the accelerometer, it does give me this warning now, though:
[image]

alanhortz, 2016-11-21:
@felipecocco So if I do understand you correctly, the original issue is gone but you now see a potential leak alert given by the underlying EventEmitter, is that right ?

Regarding the EventEmitter, this is something I personally encountered, but didn't had time to investigate it .... Any help is welcome ;)

I will create a new issue, with the potential memory leak ...

Alan

alanhortz, 2016-11-21:
@yowakita could confirm the same problem you encountered on side is also solved ?

Thanks,

Alan