brainexe/node-metawear #20: Toggle Accelerometer?

mgdbgj, 2016-03-14:
I'm trying to toggle the accelerometer by using the switch ...  and while I can turn it on the first press, the subsequent presses do not register it seems.

Does accelerometer turn off switch registration somehow?  The default Android app does not see to have that issue.

Thanks for all your work.

brainexe, 2016-03-14:
Can you please post your code?
It should be possible to start/stop the accellerometer with the functions start(), stop(), but i haven't tested it yet.

mgdbgj, 2016-03-14:
I wasn't sure whether unsubscribing, etc. was required too ...
`
var devices = require('./../src/device');

var rate  = parseFloat(process.argv[2]) || 50;
var range = parseFloat(process.argv[3]) || 2;

devices.discover(function(device) {
    console.log('discovered device ', device.address);

    device.on('disconnect', function() {
        console.log('we got disconnected! :( ');
    });

    device.connectAndSetup(function(error) {
        console.log('were connected!');
        console.log('Start accelerometer with ' + rate + 'hz ang +-' + range + 'g');

        var accelerometer = new device.Accelerometer(device);
        var logger        = new device.Log(device);

        accelerometer.setOutputDataRate(rate);
        accelerometer.setAxisSamplingRange(range);

        logger.startLogging(false);

        accelerometer.setConfig();

        accelerometer.onChange(function(data) {
            console.log("x:", data.x, "\t\ty:", data.y, "\t\tz:", data.z);
        });

        var broadcast = false;
        var switchRegister  = new device.Switch(device);
        switchRegister.register();
        switchRegister.onChange(function(status) {
		console.log("Switch status: ", status);
		if (status == 1) {
			if (broadcast == false) {	
				console.log("turning on");
        			accelerometer.enableNotifications();
        			accelerometer.enableAxisSampling();
			        accelerometer.start();
				broadcast = true;
			}
			else {
				console.log("turning off");
			        accelerometer.stop();
        			accelerometer.unsubscribe();
        			accelerometer.disableAxisSampling();
				broadcast = false;
			}
		}
	});

    });
});
`