brainexe/node-metawear #21: Question Regarding Heart Rate Sensor

FrenchBully, 2016-03-18:
Great stuff, and thanks for putting together such an awesome library.  I know you're in the process of expanding the register and was wondering any word on when that will happen?  I wanted to be able to add a Heart Rate Sensor, similar to the app they created [here](https://github.com/mbientlab-projects/HeartRate/tree/master/HeartRate).  I was trying to use the GPIO, but it seems a little over my head.  Any feedback and direction would be greatly appreciated.  Thanks in advance for the help.

brainexe, 2016-03-27:
Hey @FrenchBully, the [GPIO register](https://github.com/brainexe/node-metawear/blob/master/src/registers/gpio.js) is currently not really implemented

The job is basically to port the [Android implementation](https://github.com/mbientlab/Metawear-AndroidAPI/blob/04a18a8b3eb79669a3b3f96951953ff1991c26d6/library/src/main/java/com/mbientlab/metawear/impl/DefaultMetaWearBoard.java#L3986) to the [nodejs GPIO register](https://github.com/brainexe/node-metawear/blob/master/src/registers/gpio.js). But unfortunately I don't have time to do it.

The next step would be to use the analog GPIO input to read out the heart rate, like in the [example app code](https://github.com/mbientlab-projects/HeartRate/blob/master/HeartRate/HRViewController.m#L171)

FrenchBully, 2016-04-08:
Unfortunately I wouldn't know where to begin to port it over.  Anyway you could expedite the implementation of heart rate sensor?  I'm willing to compensate for your time?  Thanks again for the help.

brainexe, 2016-04-19:
Hey @FrenchBully ,
I just created a new branch, called "[gpio](https://github.com/brainexe/node-metawear/tree/gpio)" which includes an early version of the GPIO component. Right now I don't have access to a metawear device, so I have no chance for testing and finishing development.

**Open steps to do:**
Implement missing argument in callbacks for the methods [enableNotifications/readAnalogIn/readDigitalIn](https://github.com/brainexe/node-metawear/blob/gpio/src/registers/gpio.js) (see todo). I recommend to implement some logic in examples/gpio.js and add logging for the received "buffer" during development. When you call the script with DEBUG="noble-device" you can check the whole communication in detail.
With[ buffer.readInt16LE()](https://nodejs.org/api/buffer.html) you can access the data from metawear device. I haven't checked the related android code, but you can take a look [here](https://github.com/mbientlab/Metawear-AndroidAPI/blob/04a18a8b3eb79669a3b3f96951953ff1991c26d6/library/src/main/java/com/mbientlab/metawear/impl/DefaultMetaWearBoard.java#L3060). Just extract the relevant data from the "buffer" and pass it as argument to "callback". This should be only one value?! 

When the basic-GPIO functionality is working, you can start developing the logic for the pulse sensor.
[Here](https://github.com/mbientlab-projects/AndroidHeartRateMonitor/blob/master/app/src/main/java/com/mbientlabs/heartratemonitor/HeartRateSensorFragment.java) you can find the android code, which was easier to read for me :)

I guess the rough steps are:
- gpio.enableNotifications(GPIO_PIN) to subscribe pin changes
- with setInterval(50) read gpio.readAnalogIn(GPIO_PIN, gpio.ANALOG_READ_MODE.ADC, callback); you can request the current value each 50ms.
- [port callback functionality ](https://github.com/mbientlab-projects/AndroidHeartRateMonitor/blob/master/app/src/main/java/com/mbientlabs/heartratemonitor/HeartRateSensorFragment.java#L79)

But keep in mind, the GPIO branch was created in less than an hour and I have no device for testing. So i promise there are bugs in the code and it will not work out of the box. 
  
I recommend to create a fork from my branch "gpio" and commit you stuff there. Then I can take a look into your implementation and can give some advise when needed.