brainexe/node-metawear #22: Warnings/error about running as root

jwhendy, 2016-03-25:
Greetings,

Just found this from [this mbient forum post](http://community.mbientlab.com/discussion/347/python-api). I started my bluetooth service with `systemd` and successfully connected to my MetaWear CPro via `bluetoothctl`. When I run `DEBUG="noble-device" node examples/all.js`, I get these errors:

```
$ DEBUG="noble-device" node examples/all.js
noble warning: adapter state unauthorized, please run as root or with sudo
               or see README for information on running without root/sudo:
               https://github.com/sandeepmistry/noble#running-on-linux
```

If I run with `sudo` or as `root`, I get nothing. I know ~0 about bluetooth communication so I'm a bit lost about what to do! Thanks for any pointers.

brainexe, 2016-03-27:
Hey @jwhendy, what operting system are you using? And you have to ensure, no other device is connected to the metawear device at the same time.

jwhendy, 2016-03-27:
Shouldn't be anything else connected that I know of, but maybe my phone auto paired? I connect from the computer, then run the node command, right?

brainexe, 2016-03-27:
mhm...please check the [Getting-started page from noble](https://github.com/sandeepmistry/noble/wiki/Getting-started#what-you-need), maybe you have to install some missing driver or bluez?

But you can check following command, which should list your device:
```
sudo hcitool lescan
LE Scan ...
F9:D5:4F:B0:AB:3C MetaWear
```
```

jwhendy, 2016-03-28:
I have everything I think I need for bluetooth:
```
bluez 5.38-2
bluez-libs 5.38-2
bluez-utils 5.38-2
libbluray 0.9.2-1
pulseaudio-bluetooth 8.0-3
```

Interestingly, I can't seem to connect to it now! It's not even showing up with `hcitool lescan`. I can see it via "regular bluetooth" using `bluetoothctl`.

```
[bluetooth]# devices
Device C2:4E:2B:59:48:F1 MetaWear
```

```
$ sudo hcitool lescan
LE Scan ...
24:4B:03:9A:F7:C7 (unknown)
```

Not sure what that is, but it doesn't match the adress of the metawear from `bluetoothctl`. I'm stumped!

jwhendy, 2016-03-28:
Alright, got out my second CPro. Here's the process I tried:

```
$ sudo hcitool lescan
LE Scan ...
F5:EB:4F:1D:77:A1 MetaWear
F5:EB:4F:1D:77:A1 (unknown)
```

```
$ DEBUG="noble-device" node examples/all.js
noble warning: adapter state unauthorized, please run as root or with sudo
               or see README for information on running without root/sudo:
               https://github.com/sandeepmistry/noble#running-on-linux
```
 Bah. In the process of writing this, I found the linux-specific command on the noble page:

```
sudo setcap cap_net_raw+eip $(eval readlink -f `which node`)
```

After that I can run successfully. Sorry for the noise :(

Looking forward to messing around; this library is perfect! Without `DEBUG`, I get really nice data that is easily transformed into `.csv` format just like what I hoped for. The example looks like I can easily change the rate and range as well; very nice touch.

Thanks!

closed 2016-03-28 by jwhendy

brainexe, 2016-03-29:
Ah, very good!

The setcap command was already part of the [readme.md](https://github.com/brainexe/node-metawear#linux-ubuntu--raspberrypi) of the node-metawear project :)

jwhendy, 2016-03-29:
Wow. Indeed. Serious party foul on my end... Thanks for the patience and for pointing me in the right direction. Love the work you've done on this for everyone!