brainexe/node-metawear #25: range error on example

yutakau, 2016-08-21:
I'm using Raspberry Pi.

pi@raspberrypi:~/work/node_metawear/node-metawear $ DEBUG="noble-device" node examples/all.js 
  noble-device start +0ms
discovered device  f3:5b:f2:a4:19:27
were connected!
  noble-device  +1s send LED <Buffer 02 03 02 02 10 01 e8 03 f4 01 e8 03 c4 09 00 00 14>
  noble-device  +7ms send LED <Buffer 02 01 02>
  noble-device  +3ms send GPIO <Buffer 05 0b 00>
  noble-device  +0ms send GPIO <Buffer 05 0b 01>
  noble-device  +1ms send GPIO <Buffer 05 0b 02>
  noble-device  +1ms send LOGGING <Buffer 0b 0b 00>
  noble-device  +0ms send LOGGING <Buffer 0b 01 01>
  noble-device  +2ms send LOGGING <Buffer 0b 0d 01>
  noble-device  +1ms send LOGGING <Buffer 0b 07 01>
  noble-device  +0ms send LOGGING <Buffer 0b 08 01>
  noble-device  +1ms sendRead LOGGING <Buffer 0b 84>
  noble-device  +0ms sendRead LOGGING <Buffer 0b 85>
  noble-device  +2ms send DATA_PROCESSOR <Buffer 09 03 01>
  noble-device  +0ms send DATA_PROCESSOR <Buffer 09 07 02 01>
  noble-device  +1ms send MECHANICAL_SWITCH <Buffer 01 01 01>
  noble-device  +1ms sendRead TEMPERATURE <Buffer 04 81 00>
  noble-device  +4ms sendRead SETTINGS <Buffer 11 81>
  noble-device  +2ms send AMBIENT_LIGHT <Buffer 14 03 01>
  noble-device  +0ms send AMBIENT_LIGHT <Buffer 14 01 01>
  noble-device  +2ms send BAROMETER <Buffer 12 03 00 c0>
  noble-device  +1ms send BAROMETER <Buffer 12 01 01>
  noble-device  +0ms send BAROMETER <Buffer 12 04 01 00>
  noble-device  +1ms send BAROMETER <Buffer 12 04 01 00>
Firmware:  1.2.3
Hardware revision:  0.2
Battery: 100

buffer.js:582
    throw new RangeError('Trying to access beyond buffer length');
          ^
RangeError: Trying to access beyond buffer length
    at checkOffset (buffer.js:582:11)
    at Buffer.readInt32LE (buffer.js:743:5)
    at Function.Log.getLoggingTick (/home/pi/work/node_metawear/node-metawear/src/registers/log.js:237:26)
    at EventEmitter.<anonymous> (/home/pi/work/node_metawear/node-metawear/src/registers/log.js:101:31)
    at EventEmitter.emit (/home/pi/work/node_metawear/node-metawear/node_modules/eventemitter2/lib/eventemitter2.js:354:22)
    at Device._onRead (/home/pi/work/node_metawear/node-metawear/src/device.js:53:18)
    at Characteristic.emit (events.js:98:17)
    at Noble.onRead (/home/pi/work/node_metawear/node-metawear/node_modules/noble-device/node_modules/noble/lib/noble.js:290:20)
    at emit (events.js:106:17)
    at NobleBindings.onNotification (/home/pi/work/node_metawear/node-metawear/node_modules/noble-device/node_modules/noble/lib/hci-socket/bindings.js:390:8)

alanhortz, 2016-08-22:
Hello @yutakau,

The logging is not officially implemented yet, it is still under construction ...

I will check If I can reproduce the same error ... which RPI are you using RP1,2 or 3 ?

Can you also perform a `uname -m` in order to provide me the name and version of your distro ?

Thanks,

Alan

yutakau, 2016-08-23:
Hello alanhortz,
Thank you for your reply.
I'm using RP3, installed latest RASPBIAN JESSE.

pi@raspberrypi:~$ uname -m
armv7l
pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.4.11-v7+ #888 SMP Mon May 23 20:10:33 BST 2016 armv7l GNU/Linux
pi@raspberrypi:~$

Is this OK?

alanhortz, 2016-08-26:
Hi,

The function getLoggingTick(response) of the module Log.js is apparently not handling correctly the processing of the buffer, I just wrote a new test+impl, you can find that version here under the branch *logging* of [this repo](https://github.com/alanhortz/node-metawear).

For your info we did successfully used the logging lib through a container deployed over the air on the raspberry PI3 through [resin.io](http://www.resin.iol).  

Tell me if the fix helps you ...

Alan

yutakau, 2016-08-29:
Hi, 

It works fine! Thank you!

alanhortz, 2016-08-29:
Ok, please keep in mind that this is a temporary workaround that should work for a while  ... the spec should be a little bit more specific ...

brainexe, 2016-10-17:
sorry guys for merging branch alanhortz/logging too quickly. I just merged the (quick)fix into the master branch. I'll take a deeper look when I have more time

closed 2016-10-17 by brainexe