brainexe/node-metawear #28: service uuid 326a900085cb9195d9dd464cfbbae75a not found

Amanddn, 2016-09-16:
Hi,
I am running a simple code to blink led, it works fine but after running it for four five times, I'm getting the following error. I am running this code on nitrogen 3.10.17-37-boundary-4j3 device with SMP Debian 3.10.17-37.37 running in it

/home/debian//node-metawear-master/src/device.js:38
      if (err) throw err;
               ^

Error: service uuid 326a900085cb9195d9dd464cfbbae75a not found!
    at NobleDevice.notifyCharacteristic (/home/debian/node_modules/noble-device/lib/noble-device.js:112:21)
    at /home/debian/HandHygiene/C-Code/node-metawear-master/src/device.js:37:10
    at .<anonymous> (/home/debian/node_modules/noble-device/lib/noble-device.js:71:14)
    at Peripheral.<anonymous> (/home/debian/node_modules/noble-device/node_modules/noble/lib/peripheral.js:38:7)
    at Peripheral.g (events.js:286:16)
    at emitOne (events.js:96:13)
    at Peripheral.emit (events.js:188:7)
    at Peripheral.connect (/home/debian/node_modules/noble-device/node_modules/noble/lib/peripheral.js:43:10)
    at NobleDevice.connect (/home/debian/node_modules/noble-device/lib/noble-device.js:35:20)
    at NobleDevice.connectAndSetUp.NobleDevice.connectAndSetup (/home/debian/node_modules/noble-device/lib/noble-device.js:69:8)

Please help.

alanhortz, 2016-09-16:
Hello,

What are the version numbers : Hardware version, Firmware version, Model number ?

Thanks,

Alan

Amanddn, 2016-09-16:
Hi,
I'm using Metawear RPro with Firmware Version 1.2.3, Hardware Version 0.3

Thanks.

alanhortz, 2016-09-16:
Could you give me the code you run ?

Amanddn, 2016-09-16:
var devices  = require('./../src/device');

devices.discover(function(device) {

   setInterval(function() {
    device.connectAndSetup(function(error) {

	

        console.log('were connected!');

        // load some registers
        var log             = new device.Log(device);
        var gpio            = new device.Gpio(device);
        var led             = new device.Led(device);
        var settings        = new device.Settings(device);

        led.stop(true);

        // blink LED 5 times in blue
        led.config
            .setColor(led.config.BLUE)
            .setRiseTime(200)
            .setHighTime(500)
            .setFallTime(200)
            .setPulseDuration(1100)
            .setRepeatCount(5)
            .setHighIntensity(30)
            .setLowIntensity(1);

        led.commitConfig();
        led.play(true);

	setInterval(function() {
            device.disconnect(function(error) {
		console.log('were Disconnected!');
		process.exit();
        });
        }, 2000);
        
        settings.getDeviceName(function(deviceName) {
            console.log("Device name: " + deviceName);
	
        });
    });}, 5000);
});

alanhortz, 2016-09-25:
Hi,

Why do you put a timeout of 5 seconds around the connectAndSetup ?

It seems your code try to inform a user (by blinking 5 times) that a connection has been established [HYPOTHESIS # 1].

In the meantime I have the impression that you want to implement an autoreconnect ... [HYPOTHESIS # 2]

What branch are you using ? master from brainexe/node-metawear ?

Alan

Amanddn, 2016-09-25:
Hey Alan,
Actually i was just playing around by doing different things to get away with this error and that was the reason I put a timeout of 5 sec around the connectAndSetup (Also, as I was connecting and reconnecting connection again and again i wanted to be double sure that i reconnect after the previous connection gets properly disconnected)
Yes your hypothesis 1 is absolutely correct and after blinking led for 5 times I want to fetch the RSSI value and as I couldn't find any way to get the RSSI in a connected state, i'll have to disconnect the device, scan for the le devices to get the RSSI and then reconnect it, also i have to do this thing again and again. Also, please suggest if you know any better idea of performing it (in a connected state) right now i am calling a python script which is performing a system level scan (hcitool lescan & btmon).
And yes I am using master from brainexe/node-metawear. The difference in the actual line number and the line number in error (in device.js) is because I'm printing RSSI at starting too (using peripheral.rssi).

Aman

alanhortz, 2016-11-11:
Hi Aman,

Regarding the first point : 

I created a [small example]([url](https://github.com/node-metawear-playground/metawear-rssi-update)) on how to get the RSSI update without disconnecting the device. The device blinks 5 times once connected. I also included a way to filter the RSSI signal (kalman filter) in order to get rid of the noise. 

Does it fit your need ?

Alan

Amanddn, 2016-11-14:
Hi Alan,
This is exactly what I was looking for. Thanks for the code.

Regards,
Aman

closed 2016-11-14 by Amanddn

alanhortz, 2016-11-14:
Cool !

Glad I helped !