brainexe/node-metawear #30: Accelerometer/Gyro buffer hanging after initialization

adebiyi422, 2016-11-16:
Hello,

Thanks for the great tutorial, it is really helpful.

When I ran all.js, accelerometer.js and gyro.js for the first time, I had no issues at all. In fact, accelerometer.js and gyro.js were sending values at very high speeds. However, an hour later I tried them again and this is what I'm getting before it hangs:

**Accelerometer**
 noble-device start +0ms
discovered device  d4:ef:5d:73:8f:fe
were connected!
Start accelerometer with 50hz ang +-2g
  noble-device  send LOGGING <Buffer 0b 0b 00> +939ms
  noble-device  send LOGGING <Buffer 0b 01 01> +2ms
  noble-device  send ACCELEROMETER <Buffer 03 03 27 03> +0ms
  noble-device  send ACCELEROMETER <Buffer 03 04 01> +0ms
  noble-device  send ACCELEROMETER <Buffer 03 02 01 00> +0ms
  noble-device  send ACCELEROMETER <Buffer 03 0d 04 0a> +0ms
  noble-device  send ACCELEROMETER <Buffer 03 0a 00 14 14 14> +0ms
  noble-device  send ACCELEROMETER <Buffer 03 03 27 03> +0ms
  noble-device  send ACCELEROMETER <Buffer 03 07 07 30 81 0b c0> +1ms
  noble-device  send ACCELEROMETER <Buffer 03 01 01> +0ms

**Gyroscope**
were connected!
  noble-device  send GYRO <Buffer 13 03 0c 04> +1s
  noble-device  send GYRO <Buffer 13 02 01 00> +1ms
  noble-device  send GYRO <Buffer 13 03 0c 04> +1ms
  noble-device  send GYRO <Buffer 13 01 01> +0ms
  noble-device  send GYRO <Buffer 13 05 01> +0ms

all.js is still updating albeit slower, but it seems like the individual sensors are hanging. Any ideas why?

I would appreciate your feedback, thanks!!!

BTW: I'm running it on Ubuntu 16.04 LTS and a metamotion_R.

adebiyi422, 2016-11-16:
Upon further investigation, it seems like it may have something to do with the bluetooth low energy protocol and connecting and disconnecting from the device. If I perform a soft-reset through the app, it works fine again for a few minutes. Is there a way to do this through node? My application is going to require it to keep up with the initial speed.

Thanks!

alanhortz, 2016-11-17:
Hi,

Can you provide the HW version, FW version and Model number ?

**Regarding the first issue** : 

What happens if you let the device close to computer's BLE adapter more than an hour ? It may be related to the way your BLE adapter consider the connection as lost (just an hypothesis, never went trough this problem before).

**Regarding the reset** : The reset can be performed through the debug register. node-metawear/src/registers/debug.js

Alan

adebiyi422, 2016-11-17:
Hi Alan,

Thanks for getting back to me so quickly. 

My sensor is a MetaMotion-R r.01 and it had updated to the latest firmware(sorry I'm not sure which version), before it broke and stopped responding to the iOS app.

**Re the first issue:** After a while the device stopped responding even to the Metawear ios App after about 15 minutes, and I think it is broken due to the constant streaming. I have used the TI-sensortag in the past, and I have not had issues with a continuous BLE connection for up to two hours. I will say that exiting the node program with Ctrl-C does cut the BLE connection to where I have to reconnect, but your node program connects independently each time so I didn't think it was an issue.

**Re (reset):** Thanks, I will use it. I just run it with DEBUG="noble-device" node src/register/debug.js, right?

Thanks again!
Aminat

alanhortz, 2016-11-18:
Hi,

Maybe something related to the FW version it-self, try to downgrade the version. The mbientlab forum can help you with that.

For the reset feature, I suggest you run `DEBUG="noble-device" node examples/debug.js` to get an idea about how to perform a soft-reset.

I consider this topic closed since you encounter exactly the same behaviour with the iOS app.

Regards,

Alan

closed 2016-11-18 by alanhortz