brainexe/node-metawear #32: EventEmitter - Potential memory leak

alanhortz, 2016-11-21:
The following issue has been observed when the accelerometer example is executed on a MacOS device. 

`MaxListenersExceededWarning: Possible EventEmitter memory leak detected. 11 write listeners added. Use emitter.setMaxListeners() to increase limit`

Next step : Confirm the same kind of problem occurs on Linux based OS.

alanhortz, 2016-11-21:
@brainexe Did you ever encountered this problem ? Do you have some input to share ?

Thanks,

Alan

ibaseta220, 2017-01-12:
Same problem here on Ubuntu 16.04. Here is the output:

`discovered device  c7:e4:d4:f9:aa:94
We are connected!
Start accelerometer with 7hz ang +-2g
(node) warning: possible EventEmitter memory leak detected. 11 write listeners added. Use emitter.setMaxListeners() to increase limit.
Trace
at Characteristic.addListener (events.js:239:17)
at Characteristic.once (events.js:265:8)
at Characteristic.write (/home/pi/...
`