brainexe/node-metawear #42: Multiple MetaWear

nathanaellow, 2017-02-15:
Hi,

Is it possible to discover and connect multiple metawear devices at one go?

Thank you!

alanhortz, 2017-02-20:
Hi @nathanaellow 

Normally it should be ok but it still depends on the platform and software stack you run on.

Did you try something ? If yes, could you provide me the platform on which you executed your test ?

You should have no problem to create a small snippet, please share I will try on different platform and let you know my results ...

Regards,

Alan

alanhortz, 2017-02-24:
@nathanaellow 

I suggest you to follow the [issue #45](https://github.com/brainexe/node-metawear/issues/45)  to get feedback from another user that encounter the same need as you.

Regards,

Alan

nathanaellow, 2017-02-25:
@alanhortz 
Sorry for the delayed reply. I'm new to this and learning as I go. I'm currently testing on RaspberryPi using `devices.discover `. Kudos to @hamilton on finding that out. I am able to discover and connect more than 1 MetaWear now.

Also, to my understanding noble is able use multiple adapters by issuing `sudo NOBLE_REPORT_ALL_HCI_EVENTS=1 node <your file>.js`. Is it possible to establish connection between 1 MetaWear and 3 Bluetooth adapters for possible trilateration? 

Right now I am only able capture my RSSI data through scanning from only 1 adapter. I'm looking at ways to make this possible.

Nathanael

alanhortz, 2017-03-02:
@nathanaellow Thanks for the feedback, at the end I am so involved in the lib that some obvious methods to me are not properly documented for the users. 

Regarding you need, when the device is connected, normally it should not be connected to another adapter except in GUEST mode (you have to experiment it). There is another way to move forward, you can use the iBeacon mode and listen for the rssi updates.

Here is an [example](https://github.com/handsonio/metawear-indoor-location) I made with an additional kalman filter. Then you would connect to it if it is near enough.

Enjoy !

closed 2017-03-02 by alanhortz