brainexe/node-metawear #43: sensorFusion quaternion example does not produce values

hamilton, 2017-02-19:
So happy to find this library! Thank you for putting it together.

I am having trouble running the quaternion example with my Metamotion R unit. Is the code snippet in `/tests` known to work at this point? All of the other supported examples seem to work fine.

alanhortz, 2017-02-20:
Hi @hamilton,

Could you provide me the Serial Number, Hardware revision, Firmware revision, Model Number ?

On which platform do you run the example ?

I have encountered the same issue and it is usually solved with a soft reset, I advice you to use either the mobile app and the soft reset function provided.

Regards,

Alan

alanhortz, 2017-02-20:
Hi @hamilton,

I checked again and it could be something related to the way the configuration is written (in fact not written :), I figured out that the configuration of the sensor is forgotten in the example.

I suggest you the following procedure to write the proper config and then attempt again: 

1) Open the Metawear App (I have tested on IOS not yet on Android)
2) Connect to your MetaMotion R
3) Setup the Sensorfusion Mode to NDOF and output to Quaternion
4) Start streaming
5) Stop streaming
6) Disconnect
7) Launch the example again

Please let me know, in the meantime I will check again to make sure the example is properly updated with the right code or I will fix the issue with the configuration.

Alan

alanhortz, 2017-02-20:
@hamilton 

I just created a new branch regarding the configuration issue.

Would you please test this example ? https://github.com/brainexe/node-metawear/blob/issue-43/examples/sensorFusion.js

Please perform a soft reset and try again with that exact example code.

Thanks,

Alan

hamilton, 2017-02-20:
Thanks for the response Alan! The quaternion streaming w/ the NDoF alg works properly from the mobile app.

- firmware version: 1.3.3
- hardware version: 0.1
- model number: 5 (MetaMotion R)
- serial: 00F3AF

I ran the platform on Node 4.4.7, on OS X Yosemite 10.10.5.

hamilton, 2017-02-20:
@alanhortz The example provided appears to work.

alanhortz, 2017-02-20:
@hamilton 

In the meantime I will investigate what happens with the configuration.

Please doesn't hesitate to advice how you would use it ...

Thanks,

Alan

alanhortz, 2017-02-21:
@hamilton 

Glad to hear that everything seems to work fine.

I will publish a pre-release version of the 0.3.8 including the fixed example and will investigate the code to make sure the configuration does cover every sensors whatever the previous configuration could be.

I suggest you to run the same example on a different hardware, like a raspberry pi 3 for example, I have encountered slow streaming on Mac OS. 

Regards,

Thanks for your help,

Alan

alanhortz, 2017-02-23:
example fixed in the pre-release v0.3.8-alpha.1

hamilton, 2017-02-24:
Fantastic - thanks!

closed 2017-02-24 by hamilton