brainexe/node-metawear #45: difficulty discovering more than one MetaMotion R

hamilton, 2017-02-24:
Hello again -

I'm on a MBP running Yosemite 10.10.5, attempting to discover 4 MetaMotion Rs near me, so that I can attach one on each limb. I'm having trouble discovering more than one at a time, however, which limits my intended application of having these on dancers. I know they're all discoverable, since through the mobile app I can connect to each one individually. The firmware on the devices are all up-to-date.

HW versions: 0.1
FW versions: 1.3.3

Any ideas what might be causing the issue?

alanhortz, 2017-02-24:
@hamilton,

It could be an impediment with the underlying layer which is noble-device (uses noble).

I will check if I can find a more clues about what happens, maybe one node process for each sensor could do the job. I suggest you trying with independent process just to validate this hypothesis.

Regards,

Alan

hamilton, 2017-02-24:
It turns out this is not a bug, but is indeed due to the `noble-device` inheritance. The examples provided usually account for only one sensor, so using  `devices.discover` actually does not discover all available devices, but only the first one to be discovered. Using `devices.discoverAll` or any of the other discovery events [from `noble-device` here](https://github.com/sandeepmistry/noble-device/blob/master/lib/util.js) solves the issue.

I bet most people who end up using `node-metawear` are probably more like me - someone who isn't as familiar with `noble` enough to catch something like this. Maybe a new example or a small note on `devices.discover` / `devices.discoverAll` / `devices.discoverWithFilter` would clarify.

At any rate, thanks for the response! Going to close this issue.

closed 2017-02-24 by hamilton

alanhortz, 2017-02-24:
@hamilton 

I am eager to know if you succeed to stream from multiple sensors at the time / same rate. Please get back to me with the results of your test.

MacOS is known to me for having a lower transfer rate when it comes to stream (at least on my MacBook Pro), I don't know if it is due to the underlying node layer communicating with the BLE stack on the OS.

Regards,

Alan

hamilton, 2017-02-24:
@alanhortz I'll let you know when I'm back in front of my sensors. The application here is actually streaming from sensors to an independent unit run by an Intel Edison, so I'm hoping when I get to testing the streaming that there is no issue there.

This would be a great excuse to get moving on #41, if y'all have the bandwidth to start it. I bet you a number of people have tried this library without knowing about `devices.discoverAll`.

alanhortz, 2017-02-24:
@hamilton Feel free indeed, I started to experiment with JSDOC to Markdown in the sensorfusion branch [here](https://github.com/alanhortz/node-metawear/blob/sensorfusion/DOCUMENTATION.MD)