brainexe/node-metawear #47: OutputDataRate question

mdgagne, 2017-03-22:
I've noticed the acceleromter.js example supports specifying different output rates and range; however, the output rate doesn't seem to have an impact on the rate acceleromter data is posted to the console. Am I misinterpretting the meaning of this? Is setOutputDataRate known to work?

Thanks

alanhortz, 2017-03-22:
Hi @mdgagne,

I assume you just changed the rate at the script execution time and you observe the result on the console.

The lib will adjust your input to snap to the following values : 

**OUTPUT_DATA_RATE**

    0.78125
    1.5625
    3.125
    6.25
    12.5
    25.0
    50.0
    100.0
    200.0
    400.0
    800.0
    1600.0
    3200.0

**ACC_RANGE**  2, 4, 8, 16

What happens if you use edge valuse, do you see any differences ?

Pleas also keep this [issue](https://github.com/brainexe/node-metawear/issues/18) in mind ...

Alan

mdgagne, 2017-03-23:
@alanhortz 
Thanks for your response. Yes, I see that the rates and magnitudes 'snap' to a fixed set of possibilities. I also see a change to the x,y, z magnitude based on the supplied range; however, the onChange events happen much more frequently than expected.

Here is output with a slightly modified version of the example (ignores input rate, range) and uses 1.0 and 2.0. I've also included timestamps (utc, ms since 1/1/1970).  I get similar results with many different rates values.

```
discovered device  e9:d1:c1:68:0d:86
were connected!
Start accelerometer with 1hz ang +-2g
utc: 1490228497133 ix: 0.00238037109375                 y: 0.00018310546875             z: -0.06036376953125
utc: 1490228497147 ix: 0.00238037109375                 y: 0.00042724609375             z: -0.06036376953125
utc: 1490228497174 ix: 0.00213623046875                 y: 0.00018310546875             z: -0.06011962890625
utc: 1490228497175 ix: 0.00213623046875                 y: 0.00018310546875             z: -0.06011962890625
utc: 1490228497249 ix: 0.00213623046875                 y: 0.00018310546875             z: -0.06036376953125
utc: 1490228497279 ix: 0.00238037109375                 y: 0.00018310546875             z: -0.06060791015625
utc: 1490228497280 ix: 0.00213623046875                 y: 0.00018310546875             z: -0.06036376953125
utc: 1490228497295 ix: 0.00238037109375                 y: 0            z: -0.06060791015625
utc: 1490228497326 ix: 0.00213623046875                 y: 0            z: -0.06060791015625
utc: 1490228497331 ix: 0.00213623046875                 y: 0.00018310546875             z: -0.06036376953125
utc: 1490228497333 ix: 0.00238037109375                 y: 0            z: -0.06036376953125
utc: 1490228497336 ix: 0.00238037109375                 y: 0.00018310546875             z: -0.06036376953125
utc: 1490228497337 ix: 0.00238037109375                 y: 0            z: -0.06036376953125
utc: 1490228497369 ix: 0.00213623046875                 y: 0.00018310546875             z: -0.06060791015625
utc: 1490228497370 ix: 0.00213623046875                 y: 0.00018310546875             z: -0.06036376953125
utc: 1490228497384 ix: 0.00238037109375                 y: 0.00018310546875             z: -0.06036376953125
utc: 1490228497429 ix: 0.00213623046875                 y: 0            z: -0.06011962890625
utc: 1490228497430 ix: 0.00213623046875                 y: 0.00018310546875             z: -0.06060791015625
utc: 1490228497431 ix: 0.00213623046875                 y: 0            z: -0.06036376953125
utc: 1490228497432 ix: 0.00238037109375                 y: 0.00042724609375             z: -0.06036376953125
utc: 1490228497433 ix: 0.00238037109375                 y: 0            z: -0.06036376953125
utc: 1490228497444 ix: 0.00238037109375                 y: 0            z: -0.06036376953125
utc: 1490228497506 ix: 0.00238037109375                 y: 0.00018310546875             z: -0.06036376953125
utc: 1490228497507 ix: 0.00238037109375                 y: 0            z: -0.06036376953125
utc: 1490228497509 ix: 0.00238037109375                 y: 0            z: -0.06011962890625
utc: 1490228497510 ix: 0.00213623046875                 y: 0.00018310546875             z: -0.06036376953125
utc: 1490228497510 ix: 0.00213623046875                 y: 0            z: -0.06036376953125
```

alanhortz, 2017-04-04:
Hi @mdgagne, 

Sorry for the delay of my answer ...

Could you provide me the Model number, Firmware revision, Hardware revision of your device as well as your operating device ?

At this stage i have no clue about this discrepancy ...

Thanks,

Alan

mdgagne, 2017-04-22:
Model 0
Fw 1.2.5
HD rev 0.2
Running on raspian Jessie with raspberry pi 3

alanhortz, 2017-04-26:
@mdgagne 

Despite the configurable range for the Bosh accelerometer, the minimal value I suspect to be is 12.5 like mentioned the mobile app.

I did not succeed to get a accelerometer stream below 12.5.

Could you test with 12.5 and let me know ?

Thanks ...

Alan

mdgagne, 2017-04-30:
@alanhortz 
Thanks for the response. I get updates at roughly the same rate independent of what I set the output data rate to. Interestingly, while testing this I noticed that changing the output data rate also affects the accelerometer value scale that is returned. For example, running `accelerometer.js 10 2` yields different values that running `accelerometer.js 100 2`. For this, I'm using gravity as a reference.

Are you testing this on a raspberry pi? What sensor model are you using?

closed 2026-10-03 by mdgagne