brainexe/node-metawear #48: Using accelerometer and GPIO at the same time

davecrumbacher, 2017-05-12:
I have some code that sets and clears a GPIO pin every second (in a setInterval timer). While I'm doing that I'd like to also use the accelerometer (using accelerometer.onChange), and based on the values I receive (in particular whether there is enough movement), disable the setInterval timer. However, it doesn't seem to work. The accelerometer.onChange fires as expected, and the setInterval timer loops as expected, but the gpio.setDigitalOut and gpio.clearDigitalOut commands don't work within the timer block. It appears that the first gpio.setDigitalOut works, but that's it. When I comment out the accelerometer code, the timer works just fine with gpio.setDigitalOut and gpio.clearDigitalOut.

I also tried this manually using the mbientlab sample iOS app (started the accelerometer streaming, then set a GPIO pin), and it sets the pin very briefly then disconnects.

So is it possible to do what I'm trying?

alanhortz, 2017-05-12:
Hi @davecrumbacher,

I have personally never tried out the GPIO feature of the lib but glad you did (it help us move forward) ! 

I will compare in the coming week the behaviour of the iOS app versus the node.js lib and tell you about my conclusions.

In the meantime can you provide me the following info : 

- Firmware version
- Hardware model
- Revision number
- OS type
- Node.js version
- A dump of your console output when run with `DEBUG="noble-device" node ....`
Thanks,

Alan

davecrumbacher, 2017-05-12:
Thanks for the quick reply @alanhortz! Here's the info you requested:

* Firmware version: 1.2.5
* Hardware model: MetaDetect (based on MetaWearC)
* Revision number (hardware): 0.3
* OS type: MacOS and Raspbian
* Node.js version: 7.4.0 and 4.2.4
* A dump of your console output when run with DEBUG="noble-device" node ....

```
 $ DEBUG="noble-device" node app.js
  noble-device start +0ms
discovered device  XX:XX:XX:XX:XX:XX
were connected!
  noble-device  send MECHANICAL_SWITCH <Buffer 01 01 01> +5s
  noble-device  send GPIO <Buffer 05 02 01> +5ms
  noble-device  send ACCELEROMETER <Buffer 03 03 27 03> +1ms
  noble-device  send ACCELEROMETER <Buffer 03 04 01> +1ms
  noble-device  send ACCELEROMETER <Buffer 03 02 01 00> +0ms
  noble-device  send ACCELEROMETER <Buffer 03 01 01> +1ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 15 07 71 08 69 c2> +2ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 ed 06 2d 08 b5 c2> +112ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 d9 06 61 08 25 c2> +1ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 81 06 a5 08 65 c2> +246ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 b1 06 8d 08 a1 c2> +1ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 29 07 89 08 c1 c2> +123ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 c5 06 cd 08 5d c2> +1ms
  noble-device  send GPIO <Buffer 05 01 01> +15ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 1d 07 d5 08 41 c2> +109ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 05 07 69 08 5d c2> +1ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 b1 06 69 08 f9 c1> +122ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 81 06 35 08 55 c2> +0ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 95 06 75 08 45 c2> +124ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 f5 06 25 08 71 c2> +123ms
  noble-device  send GPIO <Buffer 05 02 01> +23ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 21 07 f1 07 95 c2> +101ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 fd 06 c5 08 4d c2> +0ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 9d 06 a5 08 51 c2> +124ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 c5 06 75 08 b5 c2> +0ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 31 07 55 08 45 c2> +248ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 f1 06 49 08 69 c2> +1ms
  noble-device  send GPIO <Buffer 05 01 01> +31ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 dd 06 35 08 85 c2> +91ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 3d 07 3d 08 25 c2> +0ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 c5 06 6d 08 45 c2> +124ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 0d 07 dd 08 f9 c1> +123ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 09 07 55 08 7d c2> +1ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 15 07 71 08 69 c2> +123ms
  noble-device  received ACCELEROMETER 4 <Buffer 03 04 c9 06 5d 08 2d c2> +1ms
```

It's interesting that when I did this test it seemed to work. I need to do some more research to see how reliable/intermittent it is. Thanks.

alanhortz, 2017-05-15:
hi @davecrumbacher,

As far as I understand, you succeeded to do what you wanted to do ... So I close this issue and let you come back with the same topic in case reliability is an issue.

Alan

closed 2017-05-15 by alanhortz