brainexe/node-metawear #53: Trouble with npm install of metawear-sd-javascript package on windows 10

mirandrew, 2017-10-28:
npm ERR! Windows_NT 10.0.15063
npm ERR! argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Program Files\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install"
npm ERR! node v6.11.5
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! metawear@0.0.4 install: `make OPT_FLAGS=-Wno-strict-aliasing -C MetaWear-SDK-Cpp/ -j`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the metawear@0.0.4 install script 'make OPT_FLAGS=-Wno-strict-aliasing -C MetaWear-SDK-Cpp/ -j'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the metawear package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     make OPT_FLAGS=-Wno-strict-aliasing -C MetaWear-SDK-Cpp/ -j
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs metawear
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls metawear
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     C:\Users\webinar\MetaWear-SDK-JavaScript\npm-debug.log

alanhortz, 2017-11-07:
Hi @mirandrew, 

This is not the right repository, I assume you are using the https://github.com/mbientlab/MetaWear-SDK-JavaScript

Regards,

Alan

alanhortz, 2017-11-07:
This repo is the very first "community" version of a node API for the metawear platform.

hugepage, 2018-01-10:
So , where can i find the right depository?